OpenCircuits/OpenCircuits #18: Write unit tests for everything

LeonMontealegre, 2017-10-30:
Testing is important and I hate it

benbelow, 2017-12-18:
@LeonMontealegre I've ahd a look over the project, and I'd recommend switching the build process from gulp-concat to something more modular like browserify - that'll make individual modules much more unit testable :)

LeonMontealegre, 2017-12-26:
@benbelow Sounds interesting! I'll definitely take a look at it. Thanks!

nprisament, 2019-04-16:
when do we close this issue...

juanmanual, 2019-07-16:
When testing stops being important or @LeonMontealegre stops hating it

LeonMontealegre, 2019-07-16:
Probably when the ever-growing "Unit Tests" project gets closed or we reach a certain level of Coverage.
Basically when I, the dictator of OpenCircuits, decide :)

TGCrystal, 2022-01-13:
Once front end testing is added, this issue should be edited to reflect a desired coverage percentage to be closed.