OpenCircuits/OpenCircuits #66: Encoder "input count" misleading, should be "output count"

LeonMontealegre, 2018-01-04:


LeonMontealegre, 2018-09-18:
This should be simple but it's not since we'll have to add either the ability to specify the text in the SelectionPopup or have another hidden text field that says "Output count" and then hide/show specifically for Encoder. TBD

nprisament, 2019-02-07:
Output count should be a separate field that is hidden by default because with IC and microcontrollers, the user should be able to separately define both the number of inputs and outputs of the given IC or microcontroller (probably through the selection popup, or through both the popup and through a separate ic/mc menu)

LeonMontealegre, 2019-02-07:
Yeah, makes sense to me.
@baxtea

nprisament, 2019-02-07:
maybe "input count" and "output count" could have flags in the component class where ic is set to true by default and oc is set to false by default, but they can be overwritten by children.

closed 2019-06-14 by LeonMontealegre