OpenCircuits/OpenCircuits #86: Add "NOT" option for ports that invert signal

Rylander77, 2018-08-31:


nprisament, 2019-02-07:
What is the difference between this and a not gate?

LeonMontealegre, 2019-02-07:
@nprisament 
There is no difference, but it could be useful to just easily NOT any signal on a component with just a port.
There'd need to be some sort of UI to signal the change though

nprisament, 2019-02-07:
What I think this could be is the ability to insert a not gate into a wire, maybe if you click on a wire, you would have the option to add a not gate in the chain. However, I think that making inputs notable would make the UI a bit cluttered.

LeonMontealegre, 2019-02-07:
Adding an entire gate into it is even messier imo.
I envision something where you select a port and check a checkbox that says 'not' or 'invert' and that specific port goes from
[image]
to
[image]

nprisament, 2019-02-07:
that's an interesting idea, and cleaner than I was thinking, my one concern would that it would make it harder to understand what is happening in a given circuit by looking at it at first glance because there could, for example, be five demux and each of them could operate entirely differently based on the inversion of their ports.

KevinMackenzie, 2021-04-09:
I think its important that this is implemented per-bus to address the issue of flipping a single port making it hard to understand what's going on.  For example, a demux would have the entire output bus inverted or not, the entire select bus inverted or not, the entire input bus inverted or not, etc.  The right answer would depend on the component, and it might make sense to prevent double-negatives from flipping both the input AND output bus.

LeonMontealegre, 2021-07-14:
Decided on: Allowing both input and output ports to have this feature. The fact that it may be confusing can be blamed on the user as it is always possible to just use a NOT gate and this is just for convenience-sake.