OpenCircuits/OpenCircuits #103: "S-R Flip Flop" is actually a Latch

Rylander77, 2018-09-20:
Latches (with enable) are designed to respond to a logic high state of the enable, which is what the current SR FF does. 

A D latch using this current "SR FF" is demonstrated as follows:
[image]
which demonstrates properties properties of a latch.

Additionally, this S-R latch does not exhibit meta-stability: as shown here:
[image]

Rylander77, 2018-09-20:
A FF exhibits edge triggering: meaning at one "edge" of the clock signal (going from high to low or low to high) is when the data input can be interpreted. This is much more accurate from a timing standpoint, so long as your clock is not faster than the delay of a gate, as a cascaded NOT gate and a second latch are used to achieve positive or negative edge triggering. 

A poisitive edge triggered D FF can be seen below:
[image]

The data input (just D, the switch) only affects the state of the output LED at the positive edge of the clock, when the clock goes from low to high.

Schmaj, 2018-09-25:
Changed to SR Latch. Will implement SR FF later.

PR #112 (2018-10-02): Flip Flop -> Latch, D Latch
Changed Flip Flops to Latches in ioobjects.
Created D Latch.
Fixed gulp and item nav files to support latches.
Removed Flip Flop files.
Fix #103 
Fix #88
closed 2018-10-02 by LeonMontealegre