OpenCircuits/OpenCircuits #104: Allow any port to be tied to any other port

Rylander77, 2018-09-20:
Can yield simpler circuits

LeonMontealegre, 2019-10-24:
Okay so I've accidentally made this a feature but we need to discuss whether or not we actually want this.

At the moment it's now possible to drag an input connection onto the Node to connect it.

You can't click on the Node and connect to something else though, we may want this using something like "OPTION+Click" to do this but idk.

@KevinMackenzie, @Rylander77 what do you think?

Previously, we've enforced Nodes to only have exactly 1 input and exactly 1 output at all times.
Now it's possible for multiple outputs.

This also creates problems where if you attempt to delete 1 of the output wires, it deletes the entire path of wires.

Maybe only delete the path if you delete the last outgoing wire?

Illustration:
[image]

KevinMackenzie, 2019-10-25:
I think connecting multiple wires to one node makes sense and is desirable.  

Selecting any segment of a wire and pressing `delete` should only delete the selected segment(s) and the optionally the resulting orphans (wires that cannot reach any component), but maybe provide an option here where you can chose to also delete half-orphans (only one component reachable from it).

There should also be an easy way to select
- all reachable wire segments that are electrically equivalent, 
- all wire segments reachable from selection point via nodes of degree 2 
- maybe: the shortest path of wires between two nodes