OpenCircuits/OpenCircuits #168: WirePorts

LeonMontealegre, 2019-02-05:
Re-add "WirePorts".

Basically, they are very simple Components that are used to split wires and are rendered as a single port.

Example:
[image]

Try to write simple unit tests before implementing them to make sure they act appropriately.

Functionally, they act just like BUFGates.

For implementing them they'd be Component with exactly 1 input and output.

For rendering, you'd probably want to just put an if-statement before the rendering of the IOPorts in the ComponentRenderer.ts checking if the object is a WirePort and if it is then have a specific rendering process for them and 'return' early.
Or, potentially a better solution would be to set the origin/target of each input/output port to be at the origin of the local transform and the rendering of the port should just automatically happen but it could cause some rendering issues.

nprisament, 2019-02-12:
#177

LeonMontealegre, 2019-02-12:
Don't close issues until the PR has been merged please

nprisament, 2019-02-12:
> Don't close issues until the PR has been merged please
got it

PR #177 (2019-02-15): Wireport
Adds wireports to the refactor

does not implement inserting wireports, deleting wireports, or reassigning wireports.

Fix #168
closed 2019-02-15 by LeonMontealegre