OpenCircuits/OpenCircuits #175: Wire Utility with WirePorts

nprisament, 2019-02-05:
When a WirePort is clicked on, the wire utility should not allow the user to change its input or outputs.

LeonMontealegre, 2019-02-07:
Not sure what you mean by this?

nprisament, 2019-02-07:
whenever a normal output is clicked on, a wire can be made from it, whenever a wireport is clicked on, the user should not be able to make a new wire from its output to another input. This will have to be a part of the wire utility that works slightly differently if the output is on a wireport.

LeonMontealegre, 2019-02-07:
See #104 we might want that

nprisament, 2019-02-07:
ok, if that is wanted, it should be fine by default, altough it could make deleting wireports a bit trickier (unless we stick with the delete the entire wire whenever a wireport is deleted, or delete the entire wire if the wireport's output is connected to multiple inputs). For this type of behavior, wireports are probably a better solution than making a separate signal splitter component.

nprisament, 2019-02-16:
I'm gonna close this for now, because it seems like this is desired behavior

closed 2019-02-16 by nprisament