OpenCircuits/OpenCircuits #176: Update the HTTPS certificate for the host server

nprisament, 2019-02-07:


LeonMontealegre, 2019-02-07:
Not really update, more like get a certificate.
This is definitely not urgent as no actual data is being sent or taken from the server but this is definitely a goal when we make user accounts and actually send information.

nprisament, 2019-02-07:
I put in the urgent tag because insecure sites often cause users to leave the site faster than they would otherwise and it could discourage people from using opencircuits who might have otherwise.

LeonMontealegre, 2019-02-07:
A lot of sites are without HTTPS certificates as it's unnecessary if no data is being transferred around. Since all it does is encrypt and protect data being sent. So chrome or other browsers won't warn you about it unless you start trying to send some data somewhere so it shouldn't be that big of a deal.

nprisament, 2019-02-07:
chrome currently marks the site as "not secure" despite not sending/receiving server data

LeonMontealegre, 2019-02-07:
Yeah, but it shouldn't be popping up in your face and warning you about it right? It's just in the top left corner?

nprisament, 2019-02-07:
that's true, it's not blaring by any means, but it is still there.

LeonMontealegre, 2019-02-07:
Yeah, but SSL certificates are expensive so we'll try and push this off for as long as possible

nprisament, 2019-02-07:
thats fair

LeonMontealegre, 2020-01-30:
GCP gives out free certs! WHOOO FREE CERTS FOR EVERYONE!

closed 2020-01-30 by LeonMontealegre