OpenCircuits/OpenCircuits #190: ICs

LeonMontealegre, 2019-02-17:


nprisament, 2019-02-18:
I think this is the same as #63

nprisament, 2019-02-18:
Also, one interesting problem is figuring out how to identify ICs in the system as to prevent something like having an IC within itself or two ICs that are codependent

LeonMontealegre, 2019-02-18:
@nprisament 
Nope #63 is about exporting ICs and is a way long term thing - this is about reimplementing ICs in the refactor which is what I'm currently working on.

Also, ICs can be put inside other ones and that's totally okay so I don't think there's much of a problem. When you create an IC you have to select the components that are being used within it so you can't use one within itself because it already needs to be there and then a new one is created.

PR #198 (2019-02-28): Integrated Circuits
Added the basis of Integrated Circuits back in.

Currently, no way to design or create them but they should work in theory.

More tests are needed for complex circuits.

Also created a bunch of Component Utilities for helpful gatherings and collections of data.
We can now also turn any group of components into a Graph for some nice data analysis.

Fix #190
closed 2019-02-28 by LeonMontealegre