OpenCircuits/OpenCircuits #191: Delayer/repeater

nprisament, 2019-02-21:
When making a complex clocked system, users should be able to add a delay to the propagation of signals through a given wire. The component would behave like a buffer gate but it would have an adjustable time delay to the signal changes by making an asynchronous promise to change the state whenever it is activated.

LeonMontealegre, 2019-02-21:
Lmao - also not paying attention in physics I see.

But yeah - that seems like a feasible and somewhat easy change. But requires a bit of UI and some internal changes so I wouldn't mark it as small

nprisament, 2019-02-21:
The internals can function the same as any other component with the activate function making a promise and then returning. The UI would be the same as the clock, so it is just a matter of that UI being finished.

LeonMontealegre, 2019-02-21:
It would require passing a variable to the Propagation request and also storing that variable somewhere. Thinking about it - it wouldn't be very easy because the way propagations are eventually executed is that it collects requests over time and sends a bunch at once so that things don't go out of sync so the custom delay would have to be some large amount. I guess it could multiply to the default propagation time. Yeah actually that would work I think.

Honestly, we're mostly just waiting on the Popup UI.
And this is pretty low priority but a good idea.

nprisament, 2019-02-22:
couldn't it work something like:

`public activate() {
	setTimeout(function(){
                super.activate(this.inputs[0].getIsOn());
        }.bind(this), this.delay);
}`

LeonMontealegre, 2019-02-22:
It would be a terrible idea to have each component do their own setTimeout. That would lead to small inconsistencies in thread sleeping and having propagations all out of sync. I think I know how to do it and I'll get to it eventually.

OwenMcGee, 2021-10-29:
I would suggest rather than adding a new component to do this, simply adding a variable "delay time" input to BUF gates.