OpenCircuits/OpenCircuits #212: More build options

LeonMontealegre, 2019-03-09:
Right now running `npm run build` can take a while as it has to compile all the TS files into JS.
It'd be nice to have the option to just copy certain non-ts files (from app, data, public, etc.) to the build folder with something like:
`npm run build -- --copy`
or specify a file or folder like:
`npm run build -- --copy="PATH"`

It would also be nice to be able to just compile a certain ts file but I'm not sure if that's possible as they are all bundled together but if a 'dev' toggle could be applied that has each ts file compiled to its own separate js file then this could be possible.
`npm run build -- --dev --file="site/public/ts/A_FILE.ts"`

This might require moving away from Webpack but hopefully, there's a way to do it.

Also the ability to minify/uglify the final code for production would be very helpful
`npm run build -- --production`

Here's some code I managed to put together that copies specific files/directories but it gives an error (but works).

`
module.exports = (env, argv) => {

    if (argv.copy) {
        if (argv.copy === true)
            return strictCopy;

        var path = argv.copy;

        if (!path.startsWith("site/"))
            throw new Error("Cannot copy files outside of the `site` directory!");

        // Strip 'site/' from beginning of path
        var stripped = path.substring("site/".length());

        // Strip 'public/' from path
        if (stripped.startsWith("public/"))
            stripped = stripped.substring("public/".length());

        if (stripped.startsWith("ts/"))
            throw new Error("Cannot strictly copy TS files! Must compile first!\nRun `npm run build` without arguments!");

        strictCopy.plugins = [new CopyWebpackPlugin([{ from: path, to: stripped }])];
        return strictCopy;
    }

    if (argv.mode === 'development') {
        // do some different stuff maybe
    }

    return defaultConfig;
};
`

where 

`
var strictCopy = {
    output: {
        path: path.resolve(__dirname, 'build')
    },
    plugins: [
        new CopyWebpackPlugin([
            { from: 'site/public/index.php', to: 'index.php' },
            { from: 'site/app/',  to: 'app/' },
            { from: 'site/data/', to: 'data/' },
            { from: 'site/public/css/', to: 'css/' },
            { from: 'site/public/img/', to: 'img/' }
        ])
    ]
}
`

nprisament, 2019-03-10:
For uglifying, there is an npm module called uglify.js that uglifies the code that webpack spits out, it is terrible for debugging, but would be good for production.

nprisament, 2019-04-17:
https://www.npmjs.com/package/uglify-js

PR #581 (2021-02-19): The Reactor
# THE REACTOR
## AKA, the, frontend, React Refactor

This humungous PR is a major refactor for OpenCircuits going forward.
It is a complete redo of the frontend code to use the lovely framework, React.

Other features include:
 - A dev server that auto-reloads the page whenever a file is changed for a better development experience
 - A new and improved file structure for the project
 - A ton of new bugs!
 - The beginnings of great, new documentation
 - Much, much, much better and safer code

### TODO
- [x] IC designer/viewer
    - [x] IC designer tool
        - [x] resize and port- [... cut]
closed 2021-02-19 by LeonMontealegre