OpenCircuits/OpenCircuits #233: "Are you sure you want exit" Popup

LeonMontealegre, 2019-03-15:
For unsaved circuits

nprisament, 2019-03-19:
there seem to be two primary ways to implement this, either a mouseout event (the mouse left the window) or an onbeforeunload event (but this is known to be disabled by default in some browsers, eg firefox) on the document / window. Alternatively, we could use both of these methods.

nprisament, 2019-03-19:
The popup could be a view like the loading screen

LeonMontealegre, 2019-03-19:
This has already been implemented

// Prompt for exit
window.onbeforeunload = function(e) {
    if (!saved) {
        var dialogText = "You have unsaved changes.";
        e.returnValue = dialogText;
        return dialogText;
    }
};

Just gotta find a place to put it and store the "saved" variable. Probably in Exporter idk.

PR #285 (2019-04-17): Exitpopup
Fix: #233 

This implements 
- dev vs prod build modes
- global constants passed from the build mode
- global variables in the utilities
- the exit popup that is hooked in the Main.ts as a promise
- the SaveAction action
- setting the global SAVED value from the ActionManager appropriately

NOTE: _Does not implement a saving mechanism or a way to excecute the SaveAction_
closed 2019-04-17 by LeonMontealegre