OpenCircuits/OpenCircuits #238: Selection of Ports

LeonMontealegre, 2019-03-18:
Ports should be allowed to be selected through a selection box

[image]

nprisament, 2019-03-19:
would this apply to the ports on a wireport?

LeonMontealegre, 2019-03-19:
@nprisament  Wireports don't have visible ports (or at least they shouldn't). They're objects that can be normally selected.

nprisament, 2019-03-19:
@LeonMontealegre with my implementation right now, they are just an input and output port that are rendered on top of each other. I am pretty sure this is how they were implemented before. They aren't a lack of port, they are entirely port (although they should still be selectable like a component right now).

PR #276 (2019-04-10): Port selection
Fix #238 

In selection tool, I added an addPortSelection method. It mirrors addSelection, except addSelection only takes ioobjects and ports are not ioobjects. Then, in activate, we only add ports to our selections if there are not other non-port objects already selected (as per your preference).

Changed the render method in component renderer to also render selected ports, as well as other objects.

IOPortRenderer has considerable edits because we condensed some duplicated code. All other file edits are shallow modifications to match changed method args etc.

Continued (poten [... cut]
closed 2019-04-10 by LeonMontealegre