OpenCircuits/OpenCircuits #357: Add "ComponentGraph" class

LeonMontealegre, 2019-07-23:
Or something - basically we need a wrapper class to make interfacing a component graph and regular graph easier and simplify the code involving graphs.

Robert-Hammond, 2020-08-18:
What is a component graph?

LeonMontealegre, 2020-08-18:
Right now, the way to represent a circuit as a Graph is by having the indices of components and their wires in the graph, which makes it a bit abstract to get useful information out of it.
It would be nice to have a `ComponentGraph` class that acts as a less abstract interface between the idea of a Circuit and a Graph representation using only numbers.

I'll more precisely lay out what exactly it should have to be helpful in the future

closed 2025-07-06 by LeonMontealegre