OpenCircuits/OpenCircuits #368: Create General Tour

LeonMontealegre, 2019-08-09:
The goal is to have something similar to Desmos where the user has to do certain tasks to move onto the next thing and it's a nice little introduction to OpenCircuits.

Should be something like:
1. Welcome to OpenCircuits!
2. Let's begin by adding an object to the circuit (a Switch): Click and drag an object onto the grid to do this
3. Great! Now let's add another one.
4. Super! Now let's add an ANDGate.
5. Awesome! Finally let's add an LED!
6. Now that we have these objects let's hook them together! Start by clicking on the Switch's Port and then drag to the ANDGate's port (highlight them)
... Wire up the whole circuit
then Press the Switches to turn them on and watch the LED turn on!!!!

This is probably gonna be a pretty intensive issue since intro.js simply highlights things.
It's gonna require a lot of very annoying and specific code for only allowing interaction with one thing at a time and calculating the correct positions for things to highlight them and checking when a person did them.

Annoying.

LeonMontealegre, 2019-08-12:
We might wanna look into this
https://github.com/xbsoftware/enjoyhint

TGCrystal, 2021-05-25:
Looking into this
[https://github.com/elrumordelaluz/reactour](https://github.com/elrumordelaluz/reactour)
and this
[https://github.com/gilbarbara/react-joyride](https://github.com/gilbarbara/react-joyride)

LeonMontealegre, 2021-05-25:
@TGCrystal You might want to look at this branch: https://github.com/OpenCircuits/OpenCircuits/tree/MainTutorial

They made some pretty decent progress, but it is Pre-React so who knows