OpenCircuits/OpenCircuits #383: Docker / headless usage is dependent on "headded" features

Rylander77, 2019-08-20:
Two problems I've experienced so far when attempting to run the Docker image on top of [Proxmox VE](https://www.proxmox.com/en/proxmox-ve/) and [Docker Desktop](https://www.docker.com/products/docker-desktop) for Windows:

1. The running command when starting a container is `npm run start`,
https://github.com/OpenCircuits/OpenCircuits/blob/f56b12139e50c7d7bedff35033ec7174d6c09347/Dockerfile#L31
 which calls the script: `cd ./build && opener http://localhost:8080 && ./server` 
https://github.com/OpenCircuits/OpenCircuits/blob/f56b12139e50c7d7bedff35033ec7174d6c09347/package.json#L44
[Opener](https://www.npmjs.com/package/opener) is OS dependent, but only functions to also open up a browser to the page, not needed in a headless run. The specific part of `opener.js` of note is found [here](https://github.com/domenic/opener/blob/3deb561d900a268173b79eec0126af30e16d4b4b/lib/opener.js#L17-L30), where `xdg-open` is called when the platform is determined not to be Mac or Windows. This calls on the desktop environment to open the file or URL with a configured default program, or some other default behavior, both of which are meaningless within the context of headless use. Presumably an [error code](https://linux.die.net/man/1/xdg-open) is returned, which `opener.js` [just throws](https://github.com/domenic/opener/blob/3deb561d900a268173b79eec0126af30e16d4b4b/bin/opener-bin.js#L6-L10) without a ton of information:
```
> opencircuits@0.0.1 start /go/src/github.com/OpenCircuits/OpenCircuits
> cd ./build && opener http://localhost:8080 && ./server

/go/src/github.com/OpenCircuits/OpenCircuits/node_modules/opener/bin/opener-bin.js:8
        throw error;
        ^

Error: spawn xdg-open ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:240:19)
    at onErrorNT (internal/child_process.js:415:16)
    at process._tickCallback (internal/process/next_tick.js:63:19)
    at Function.Module.runMain (internal/modules/cjs/loader.js:834:11)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)
```
Regardless, simply changing the Dockerfile to execute `cd ./build && ./server` at start allows the image to start.

2. The server is configured to listen and serve requests on loopback / `localhost` / `127.0.0.1`:
https://github.com/OpenCircuits/OpenCircuits/blob/f56b12139e50c7d7bedff35033ec7174d6c09347/site/go/main.go#L47
The loopback address is useful when running the project locally on the same host that will access the server, but does not let other hosts (outside of the `127.0.0.0/8` loopback network) access the server.  Attempting to connect to a server running on another host (physical or virtualized) host fails because the server does not respond to traffic outside of it's very own, even when a host attempts to connect to it's own server using its own IP address. Changing to ` router.Run("0.0.0.0:8080")` allows connectivity between hosts, as serving on `0.0.0.0` runs the server on all IPv4 addresses of a given host, including serving on `localhost` for the local host.

LeonMontealegre, 2019-08-20:
beeg yoshi

Rylander77, 2019-08-22:
3. Dockerfile does not build css. Fixed with adding `RUN npm run build:css` to the Dockerfile.

commit 3d3a2217 (2019-08-22): Merge pull request #384 from OpenCircuits/Issue383Branch

Fix #383
closed 2019-08-22 by LeonMontealegre