OpenCircuits/OpenCircuits #402: Button Components reset to off when copy-pasted

baxtea, 2019-09-20:
Should be an easy fix, just add a bit of internal state (currently, buttons are stateless).

baxtea, 2019-09-20:
Former comment was misinformed; buttons inherit state from `PressableComponent`. I fixed the error though, which was `PressableComponent::copy` not copying that state. Ran into propagation errors because the way propagation works is kind of insane? But fixed those too. I may try and draft a rewrite to the propagation system in the near future.

LeonMontealegre, 2019-09-20:
@baxtea
Buttons shouldn’t stay on unless being held? I don’t think I understand the problem. Also the propagation system isn’t too bad maybe it should be more clear but eh

baxtea, 2019-09-20:
My bad, meant switches, not buttons. And I'll dm you or open another issue to talk about propagation.

closed 2019-11-03 by baxtea