OpenCircuits/OpenCircuits #466: Ports snap weird with connected rows of ports

LeonMontealegre, 2019-12-07:
To reproduce:

Create an IC with ports slightly farther-apart then default for an IC

then bus the ports with an Encoder and try to snap the bottom two together.

This happens:

[image]

Robert-Hammond, 2020-08-18:
I don't see how this is a bug, it's due to the distance between ports being wider on one component than the other.

LeonMontealegre, 2020-08-18:
@Robert-Hammond yes, but intuitively, the bottom rows are going to be aligned first and therefore should be the ones that snap. But the current behavior is that it causes some sort of chain-of-snaps and then the top row is aligned.