OpenCircuits/OpenCircuits #470: IC Cullbox on Creator is incorrect

LeonMontealegre, 2019-12-10:


Robert-Hammond, 2020-08-18:
Can you describe the problem in further detail please?

LeonMontealegre, 2020-08-18:
To reproduce:
1) Turn on `DEBUG_CULLBOXES` by setting it to true in `app/core/ts/utils/Constants.ts`
2) Create IC-able circuit:
[image]
3) Create IC:
[image]
4) Resize IC
[image]

Notice how the cullbox (highlighted in green) is incorrect for the re-sized IC.

However, when created, it is correct.
[image]

PR #906 (2021-11-21): Bunch of IC Fixes
Fixes #644 
Fixes #639 
Fixes #470 
Fixes #712 
Fixes #888 
Fixes #456 
Fixes #907
closed 2021-11-21 by LeonMontealegre