OpenCircuits/OpenCircuits #528: Truth Tables

Robert-Hammond, 2020-06-22:
This is a rather big enhancement, as there are many subtasks involved. Also, it seems like a very difficult task.

**From circuit to table**
ICs will have a button that can generate/view the truth table for that IC. It will work well for ICs because they have an explicitly defined set of inputs and outputs. Some flip-flops and latches have truth tables that depend on the previous state of inputs and/or outputs (i.e. D Flip Flop)

**From table to formula(s)**
Truth tables can be simplified, and with each output there is a logical formula that can also be simplified. Preferably the truth table and formulas are either displayed simplified by default or the user has the option to simplify them.

**From formulas and tables to circuits**
Ability to create an IC from a user-created truth table.

OwenMcGee, 2021-11-19:
I'll note that the circuit to table functionality in particular may have some difficult design questions to consider. Namely the fact that some circuits which hold states in memory (think RS NOR Latches) may produce different outputs for the same input.

TGCrystal, 2021-11-21:
> I'll note that the circuit to table functionality in particular may have some difficult design questions to consider. Namely the fact that some circuits which hold states in memory (think RS NOR Latches) may produce different outputs for the same input.

If I'm not mistaking, this can be sidestepped by converting the circuit into a graph and then checking if that graph is acyclic. Only acyclic graphs could then be converted into a truth table.