OpenCircuits/OpenCircuits #575: Contextual menu copy causes error

GarrettShorr, 2020-12-15:
**Describe the bug**
Right clicking on anything (the background or an object) and selecting copy causes a javascript error. After this occurs, opening a circuit from a file no longer works.

 - OS: windows 10
 - Browser: chrome
 - Version: 87.0.4280.88

[image]

LeonMontealegre, 2020-12-15:
Issue is the with the if-statements on 
https://github.com/OpenCircuits/OpenCircuits/blob/master/site/public/ts/shared/controllers/ContextMenuController.ts#L47
and
https://github.com/OpenCircuits/OpenCircuits/blob/master/site/public/ts/shared/controllers/ContextMenuController.ts#L67
As they don't check if `navigator.clipboard` exists before `navigator.clipboard.readText`

LeonMontealegre, 2021-07-03:
Should be fixed along with #609