OpenCircuits/OpenCircuits #590: Zooming is very messed up

LeonMontealegre, 2021-02-26:
Zoom in/out, for some reason the grid zooms at a different speed from the objects.
No idea how this managed to happen. Lovely.

Xystem4, 2021-02-27:
Note that the zoom is only broken vertically, horizontally the gridlines stay in the same relative positions to the components.

LeonMontealegre, 2021-02-27:
Yeah, we figured out the problem is 
https://github.com/OpenCircuits/OpenCircuits/blob/15eccc57299f5fb25b2630b2d499baed5ba62bc0/src/site/pages/digital/src/containers/MainDesigner/index.tsx#L48

`        camera.resize(w, h); // Update camera size when w/h changes`
should be 
`        camera.resize(w, h - HEADER_HEIGHT); // Update camera size when w/h changes`

So that the camera accounts for the height of the header

PR #603 (2021-03-06): Fixed camera zoom issue
Closes #590
closed 2021-03-06 by LeonMontealegre