OpenCircuits/OpenCircuits #595: Rotating groups of components causes drift towards center of group

Xystem4, 2021-02-27:
When rotating any group of components, all components will gradually drift towards the center of the red rotation circle (with and without shift being pressed), regardless of direction rotated. Undoing will undo the rotation, but not the drift towards the center.

Components before rotation:
[image]

Components after rotating, and then undoing rotation:
[image]

This has the add-on effect that when splitting a wire and then rotating a group of components that includes both original ends of the wire, the components will drift together. Then, when undoing, the rotation will be undone and so will the splitting of the wire, but the locations of the components will not be the same as when they began. Upon un-splitting, the wire will jump back to its original location prior to being split, which may no longer be connected to (or even near) the nodes it is meant to connect to. Note that this does not cause a crash or anything besides a visual error, which is fixed upon moving the wire itself or either component (or node) that it is connected to.

Disconnected wire:
[image]

LeonMontealegre, 2021-02-27:
This is probably partly due to #590, the camera offset probably causes the rotation math to be funky since it depends on the camera. Also sounds like the components `onTransformChange` might not be firing when undoing these actions

LeonMontealegre, 2021-03-06:
It's not due to #590 :(

LeonMontealegre, 2021-03-06:
I figured it out! Apparently, the midpoint of the objects are changing as they rotate (which shouldn't happen and I'm looking into that now), but so saving the original midpoint as a state variable `onActivate`, and using that consistent midpoint value fixes it.

LeonMontealegre, 2021-03-06:
Got it! It's because
https://github.com/OpenCircuits/OpenCircuits/blob/0336e5f31b831a1685b5793336058d121dec0d28/src/app/core/tools/RotateTool.ts#L89

Is a LOOP! And on each iteration, only the first few objects have been moved; yet each time the selection's midpoint is recalculated! Meaning that it's not calculating correctly since not all objects have been moved!

PR #604 (2021-03-06): Rotation drift fix
Closes #595
closed 2021-03-06 by LeonMontealegre