OpenCircuits/OpenCircuits #624: Clicking a node already connected to a wire both starts a new wire and selects the existing wire

Xystem4, 2021-03-26:
**Describe the bug**
If a component has a wire leading out of it and into some other component, clicking that output node (to generate a new wire) will still begin the new wire, but will also select the existing wire.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a switch and two LEDs to the canvas.
2. Connect the switch to one of the LEDs.
3. Click the output node of the switch
4. The beginning of a new wire will form, but the existing wire will also be selected.

**Expected behavior**
The existing wire should not be selected.

**Screenshots**
(excuse the invisible mouse)
Circuit before clicking
[image]

Circuit after clicking
[image]

Expected behavior after clicking
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: chrome

**Additional context**
This does not occur if clicking the very edge of the node, leading me to believe that it only happens when clicking a location that has a wire hidden beneath it. I suspect this is either a result of the recent PR #593, but (far) more likely from issue #621.

OwenMcGee, 2021-09-24:
I can no longer reproduce this with just wires, but I _can_ reproduce it with other components.

Here is an example, where a constant high has its output port beneath a constant low, and clicking the area where the node is will both select the constant low and create a new wire originating from the constant high
[image]
.

mercuryhg31, 2021-10-01:
Noting that this happens with these too:
[image]

mercuryhg31, 2022-03-15:
Both on opencircuits.io and the nightly site, I can still see this issue present:
[image]

PR #947 (2022-01-14): Changed default tool to trigger after other tools have been activated
Fixes #624