OpenCircuits/OpenCircuits #638: Item preview not appearing when dragging from itemnav

Xystem4, 2021-04-15:
**Describe the bug**
When dragging a component from the itemnav onto the canvas, the item preview visual no longer displays.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to nightly.opencircuits.io
2. Open the itemnav
3. Drag any component onto the canvas
4. See no preview following cursor

**Expected behavior**
The item preview visual should display following the cursor

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: chrome

**Additional context**
I am almost certain this was introduced by pull #627

TGCrystal, 2021-04-15:
If this gets fixed (and someone has the relevant setup: Linux bare metal install, nvidia graphics card, proprietary nvidia drivers, chrome) then see if #592 is still an issue.

Xystem4, 2021-10-15:
If someone fixes this, they should take into account the new feature of placing multiple of a component simultaneously  (from PR #650 for those unfamiliar). Currently, clicking a component n times and then dragging it onto the canvas will place n of them (as if you'd just clicked on the canvas and not dragged). Assuming this is intentional, the preview should show n components when the user drags, not necessarily just 1.

PR #790 (2021-11-05): Item Nav Drag Preview
Adds a preview for ItemNav Drag'N'Drop
Also implements the 'Smart Place' feature which can be activated by holding ALT/OPTION while dropping a component from the ItemNav

Fixes #638 
Fixes #689
closed 2021-11-05 by LeonMontealegre