OpenCircuits/OpenCircuits #669: Documentation

LeonMontealegre, 2021-07-02:
Proper documentation is desperately needed, I am currently working on it but it should be officially documented (pun intended) as well.

Tasks:
- [ ] Create site
- [ ] Auto-generated JSDocs
- [ ] Full "wiki" copy-over
- [ ] Flow diagrams for the project
- [ ] Style Guide
- [ ] Explanations for larger systems like Actions, Tools, Redux, etc.
- [ ] API documentation
- [ ] Stream-line publishing the doc-site (Github Actions)
- [ ] Design Docs

TGCrystal, 2021-07-07:
Another potential type of documentation to consider adding: some kind of "exit docs" that someone could fill out if they are in the middle of working on something, but the semester is about to end and they don't plan on continuing working on OpenCircuits after that. They could explain what branch the work is on, what work has been done, and what work is still left to do.

TGCrystal, 2021-10-02:
I noticed while creating jsdocs in #630 that the _throws_ and _see_ tags do not show up in the auto-generated jsdocs. Not sure if that was an intentional omission, I messed something up/didn't see it, or it wasn't yet implemented, but I think those should also should show up in the generated documentation.

LeonMontealegre, 2021-11-04:
Note for myself:
Could be cool to have any documentation-changing PRs get published onto a sub-url of the docs site like: `docs.opencircuits.io/pull/###` so that it can be viewed without having to manually checkout and run the docs server

TGCrystal, 2021-11-12:
- [ ] Add support for tuples (see [here](https://github.com/OpenCircuits/OpenCircuits/blob/26eb8b1c9ac0a014cb27d764df0a9c661ad1263f/scripts/docs/parse.ts#L41))