OpenCircuits/OpenCircuits #724: Better Labels

LeonMontealegre, 2021-10-01:
Labels need to be improved, here are some suggestions (in general order of priority) - please feel free to add more!

- [ ] Ability to have multiple lines of text
- [ ] Font size change

Really it would be nice if labels had a whole editor type thing like typical text editors, even a markdown editor.
There's a good chance that React libraries exist for a markdown editor component. This would obviously require some tremendous changes but it's good to throw out there. This would require Labels to be pure React components most likely instead of "virtual" canvas-drawn components which might have its own slew of issues.

OwenMcGee, 2021-10-01:
Might be nice to have an option for labels attached to LEDs (I know attaching labels to components is a feature still being worked on, not yet in master) to somehow "light up" or become a brighter color or in some other way change/become more visible when the LED they are attached to becomes powered.

gshorrSPHS, 2022-04-28:
It also would be nice to have other types of decorators like horizontal and vertical lines to separate out different sections of the workspace. My terrible workaround for a horizontal line was having a label with a bunch of spaces.