OpenCircuits/OpenCircuits #842: (Error when loading) ReferenceError: gapi is not defined

TGCrystal, 2021-11-04:
Got this error once last night, just got it again now. Seems to just unexpectedly show up every now and then.

```
Auto-generated error:
Error occurred while "Rendering"
ReferenceError: gapi is not defined
at http://nightly.opencircuits.io/static/js/main.54d23cc3.js:2:224242
at Eu (http://nightly.opencircuits.io/static/js/main.54d23cc3.js:2:776544)
at e.unstable_runWithPriority (http://nightly.opencircuits.io/static/js/main.54d23cc3.js:2:813181)
at Wo (http://nightly.opencircuits.io/static/js/main.54d23cc3.js:2:717082)
at Cu (http://nightly.opencircuits.io/static/js/main.54d23cc3.js:2:776007)
at hu (http://nightly.opencircuits.io/static/js/main.54d23cc3.js:2:767502)
at http://nightly.opencircuits.io/static/js/main.54d23cc3.js:2:717305
at e.unstable_runWithPriority (http://nightly.opencircuits.io/static/js/main.54d23cc3.js:2:813181)
at Wo (http://nightly.opencircuits.io/static/js/main.54d23cc3.js:2:717082)
at Go (http://nightly.opencircuits.io/static/js/main.54d23cc3.js:2:717250)
```

TGCrystal, 2021-11-04:
Just had this happen on my phone as well, here is the error when it tries to autogenerate the issue from that link:
```
Auto-generated error:
Error occurred while "Rendering"
ReferenceError: Can't find variable: gapi
Eu@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:776195
@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:812832
hu@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:767154
hu@[native code]
@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:716956
@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:812832
Go@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:716902
Ho@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:716837
yu@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:767658
tc@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:782127
@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:649116
@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:646845
@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:645885
Promise@[native code]
gd@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:645645
@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:157619
@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:156765
@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:156251
a@http://nightly.opencircuits.io/static/js/main.ba404431.js:2:156989
promiseReactionJob@[native code]
```
If I had to guess, this is an adblocker somehow interfering with the Google authentication stuff.

LeonMontealegre, 2021-11-12:
Most likely due to GAPI script not loading before the main script (https://stackoverflow.com/questions/31144874/gapi-is-not-defined-google-sign-in-issue-with-gapi-auth2-init)

PR #920 (2021-11-30): Fixed miscellaneous small issues
Closes #538 
Closes #647
Closes #721 
Closes #842 
Closes #777
closed 2021-11-30 by LeonMontealegre