OpenCircuits/OpenCircuits #883: holding alt for pan only works for first click and drag

Kevin-771, 2021-11-10:
**Describe the bug**
While holding the alt key, the first pan via mouse works normally, but the subsequent ones don't work.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://nightly.opencircuits.io/
2. Press and hold the alt key
3. Click the left mouse button and drag to pan screen
4. Release the left mouse button, without releasing the alt key
5. Click the left mouse button again and drag

**Expected behavior**
Expected behavior should be that the subsequent click and drags should pan the canvas.

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser: Chrome

OwenMcGee, 2021-11-11:
I'm having difficulty reproducing this (also on windows and chrome). What happens for you on the subsequent drags, does a selection box appear (as would happen if you were never holding alt), or does just nothing happen?

(I'll add that I'm testing this with a touchpad, not a mouse, but I'd think I should still be able to reproduce this with my touchpad)

mercuryhg31, 2021-11-11:
It does reproduce for me (touchpad, Windows, Chrome), and yes, the selection box shows up instead of panning again.

PR #1018 (2022-04-18): Update PanTool so Works with Multiple Clicks and Drags
Resolves #883 

Change so you can continuing using PanTool until alt key is released.
closed 2022-04-18 by LeonMontealegre