OpenCircuits/OpenCircuits #897: When using the expression to circuit feature, add option to create with clocks set up as a truth table

OwenMcGee, 2021-11-17:
Clocks can be used to simulate a truth table for a circuit by having clocks connected to each input, with each subsequent clock having twice the delay of the last. This can be seen by connecting all inputs (and likely, the output as well) to an oscilloscope. An example of this is shown below (in retrospect this is a somewhat poor example as a boolean expression can only result in a single output, but just ignore that):
[image]
Or, to steal Professor Kuzmin's image to see how this oscilloscope is a truth table:
[image]

I believe it would be a useful feature to, when selecting clocks as the input component when using the expression to circuit feature, allow the user to have the clocks be created in this manner (each having double the delay of the last, and potentially connecting all inputs and outputs to an oscilloscope (or perhaps one for inputs and one for outputs, to avoid the messy connections as in my example).

I am actually hard pressed to imagine a circumstance when I would want clocks as inputs but all at the same delay and in sync, so I may even suggest making this the default behavior with clock inputs, but that's up for debate. An additional checkbox for "Create as truth table with oscilloscopes" or something is also fine.

Some issues I can predict if this was to be added:
1. There is a minimum and maximum to the amount of delay a clock can have, meaning that at most 8 clocks can be placed until it is impossible to double the delay again. I'm not sure what the best solution would be for expressions with over 8 inputs. A clock with above maximum delay could be simulated with some extra components in that section, but that might be doing too much.
2. Oscilloscopes only allow 8 inputs, so circuits with more would run into an issue trying to fit more. Although two oscilloscopes right on top of one another looks practically identical to simply having a taller oscilloscope, so that could be a way around it
3. If the oscilloscope connects both inputs and outputs, the wiring has the potential to get messy quickly, unless extra thought is put into it. Even if separate oscilloscopes are used, ideally they would be close to one another so they could actually be used to see what inputs result in what outputs.

Additional notes:
- We also need to make sure that the oscilloscope's delay isn't larger than that of the quickest clock
- The oscilloscopes being present isn't strictly necessary, and even without them this would still be more useful than the current clock inputs in my opinion

TGCrystal, 2021-11-18:
I agree that the clocks should be synchronized with offset delays like this by default. From a technical standpoint, I don't think this would be all that hard, those design issues you brought up are the hard part. Here's my thoughts:

1. Using BUFGates to delay later clocks or something like that is possible, but would be pretty hard to implement. I would prefer to just have any clock number 8 and beyond to share the same frequency.
2. Adding the second oscillscope would mean certain disaster with the placement algorithm in most nontrivial cases. Tying into point 1, if we just have all clocks number 7 and beyond (spot 8 on oscilloscope reserved for the final output) with the same frequency, then there wouldn't be a need to create a new oscilloscope since there wouldn't be any other frequencies.
3. Placement algorithm is bad, not much I can/will do about that.

Another way entirely would be to detect the number of inputs in the expression, and have a switch to connect the clocks to oscilloscopes that shows up if output is selected as oscilloscope, input as clock, and there are less than 8 inputs in the expression. May be a little harder technically, but I think it would be doable.

OwenMcGee, 2021-11-19:
I agree with everything you said, @TGCrystal!

One extra thing to add, I think it's in our best interest to (when dealing with many inputs) make the quickest delay as slow as possible, as when the delay gets as low down as 50 the delay time of the signal propagating through the circuit to the output can be significant, relative to the 50 ms. When only dealing with 7 different delay times, the largest time you can have the smallest delay be (doubling each subsequent time) is 156 ms. 150 being a nicer number would probably be good enough, as well. (Basically all I'm saying is don't start the smallest clock at 50)

Also as for your first point, I think it would be much easier with BUF gates once (_if_) someone implements the enhancement that would add a delay time option onto them. That said, I still agree with you that stopping clock delay growth after 7 inputs is a better, cleaner solution

PR #945 (2022-01-04): Expression parser improvements
closes #927 
closes #909 
closes #897 

Also includes some under the hood improvements to make more parts of the pipeline action based. This helps future proof with the proposed collaborative editing feature which requires pretty much everything to be an action. It also will make converting some of the currently frontend only code to an action which eases testability. Currently, I am waiting on an action based circuit arrangement algorithm to fully go through with that.

There are also some changes to the tree structure of the underlying algorithm. That now direc [... cut]
closed 2022-01-04 by LeonMontealegre