OpenCircuits/OpenCircuits #909: Add option to include labels when creating a circuit from a boolean expression

OwenMcGee, 2021-11-22:
When using the expression to circuit feature, all inputs are already labelled by necessity in the expression, and it could be helpful to the user to display a label next to the switches (or buttons or clocks) when they appear on the canvas. A checkbox asking "Add labels next to inputs?" on the righthand side would be sufficient.

If the "Generate into IC" option is checked, this checkbox should probably be forced to be off, since the inputs are already visibly labelled.

TGCrystal, 2021-11-23:
I like the suggestion, although I'm not sure if I agree that the option should be forced off for ICs. The labels can be useful to see when viewing the contents of the IC.

OwenMcGee, 2021-11-23:
> I like the suggestion, although I'm not sure if I agree that the option should be forced off for ICs. The labels can be useful to see when viewing the contents of the IC.

That's a good point, I suppose I'd been thinking if the option was on while the ICs option was on, it would be making the labels _outside_ the IC. I'd definitely be in favor for it making the labels appear when viewing the IC, editing the original suggestion to reflect this

PR #945 (2022-01-04): Expression parser improvements
closes #927 
closes #909 
closes #897 

Also includes some under the hood improvements to make more parts of the pipeline action based. This helps future proof with the proposed collaborative editing feature which requires pretty much everything to be an action. It also will make converting some of the currently frontend only code to an action which eases testability. Currently, I am waiting on an action based circuit arrangement algorithm to fully go through with that.

There are also some changes to the tree structure of the underlying algorithm. That now direc [... cut]
closed 2022-01-04 by LeonMontealegre