OpenCircuits/OpenCircuits #917: Add linting (eslint?)

TGCrystal, 2021-11-25:
**Is your feature request related to a problem? Please describe.**
When submitting a pull request, it is almost guaranteed that there will be some formatting issue that needs changing. Currently, most contributors will submit a pull request, Leon will point out the formatting issues, then the issues are fixed.

**Describe the solution you'd like**
Using something like eslint can not only automate this process, but also allow the linting to be run locally. Codebeat and Codacy kind of serve as linters, but are only easily useable once the pull request is actually made. eslint allows contributors to check for issues in their code before even needing to make a pull request. eslint also works with Github Actions and React, so it shouldn't be too hard to add to the current CI process.

LeonMontealegre, 2021-11-25:
Kinda a duplicate of #340 but yeah, someone is workin on it rn

TGCrystal, 2021-11-25:
> Kinda a duplicate of #340 but yeah, someone is workin on it rn

Kinda, that one is for adding more linting rules. This issue for adding linting in general.

pbabell, 2021-11-25:
Just for funsies I set up gts locally. (Hi by the way, introduced to this project by @TGCrystal). Uses eslint to follow googles ts style guide. [https://github.com/google/gts](url). Couple things like whitespace and single vs double quotes are different in this project vs what is in the guide but nothing crazy. Some other things would need to be manually changed like unused variables, and using double vs tripple equals, etc.

PR #952 (2022-01-12): Added initial ESLint implementation
Closes #917
closed 2022-01-12 by LeonMontealegre