OpenCircuits/OpenCircuits #918: Jest code coverage

pbabell, 2021-11-25:
**Is your feature request related to a problem? Please describe.**
When implementing a new feature or fixing a bug, it is helpful to know that new lines and branches are adequately tested. Without this knowledge, anyone reviewing the pull request will need to (or should) determine that the new code has adequate test coverage. Code coverage reports will improve code base robustness.

**Describe the solution you'd like**
Jest can provide code coverage reports as part of branch builds. Add a requirement for minimum new code and branch coverage. Also consider just displaying the reports and adding coverage as a requirement later.

pbabell, 2021-11-25:
Can not edit labels for some reason.. Should be under devops

TGCrystal, 2021-11-25:
This could be added in a `yarn coverage` command, similar to `yarn test`.

LeonMontealegre, 2021-11-26:
> Can not edit labels for some reason.. Should be under devops

Allowing non-members to edit labels would allow them to edit labels for all issues which I think you can see how that would be a problem. Are you able to set them when creating an issue?

pbabell, 2021-11-26:
Nope, had to leave all defaults

PR #941 (2021-12-19): Added code coverage commands
Added `yarn coverage` and `yarn coverage:all` commands, including corresponding documentation for an explanation of their use.
closes #918
closed 2021-12-19 by LeonMontealegre