OpenCircuits/OpenCircuits #1175: Header Link broken

ION606, 2022-10-13:
**Describe the bug**
When clicking the "OpenCircuits" heading, it redirects to https://www.opencircuits.io/home, which displays the message "404 page not found"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '[...](https://www.opencircuits.io)'
2. Click the header text reading "OpenCircuits"

**Expected behavior**
Opening a functional page

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Firefox, Edge
 - Version: Unsure

**Smartphone (please complete the following information):**
N/A

**Additional context**
Add any other context about the problem here.

*PLEASE MAKE SURE TO APPROPRIATELY LABEL YOUR ISSUE, YOUR ISSUE IS REQUIRED TO SPECIFY AT LEAST A GENERAL IDEA OF: Area, Difficulty, Priority, and Scope! If the issue is specific to digital/analog, please also specify that with the appropriate label!

ION606, 2022-10-13:
I am unsure of how to change the labels. Sorry....

huynchu, 2023-01-27:
I think this is fixed in model-refactor. @LeonMontealegre