OpenCircuits/OpenCircuits #1233: Analog Keyboard Shortcuts

AmolK987, 2023-03-13:
Add keyboard shortcuts to bring up common analog circuit components in order to streamline the circuit design process using OpenCircuits.

LeonMontealegre, 2023-03-13:
So what's the exact desired behavior you were thinking?

Something like:
User presses 'r' and suddenly the outline of a Resistor appears over their cursor and they can then click to place as many as they want. User can then press Escape to stop placing them?
Or does it make sense for only a single resistor place when clicked?

julianchen215, 2023-03-14:
Yeah, we were thinking about implementing it similar to how ltspice has it, where if you clicked 'r' your cursor turns into a resistor and you can place how ever many you want, and then press escape to stop placing resistors.