OpenCircuits/OpenCircuits #1280: Saving Issue

ThePackageMan, 2023-09-28:
**Describe the bug**
When a circuit is too big, it cannot be saved to the web.

**To Reproduce**
Steps to reproduce the behavior:
1. Extract and Import [CPU.zip](https://github.com/OpenCircuits/OpenCircuits/files/12753906/CPU.zip)
2. Click on 'Save'
3. Watch as the file does not save.

**Expected behavior**
Save button to disappear,
File to save to web.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Chrome OS
 - Browser: chrome
 - Version: 116.0.5845.210

**Additional context**
Downloading works, but it is a hassle to re import and overwrite older circuit files on file explorer.

*PLEASE MAKE SURE TO APPROPRIATELY LABEL YOUR ISSUE, YOUR ISSUE IS REQUIRED TO SPECIFY AT LEAST A GENERAL IDEA OF: Area, Difficulty, Priority, and Scope! If the issue is specific to digital/analog, please also specify that with the appropriate label!
IDK what to put for ^, but i'll just put Difficulty if that's right.

shusain, 2023-10-08:
Know it doesn't solve your issue but I also ran into this recently, not sure if it has something to do with the GCP account used for storage I was looking at limits here but nothing jumping out to me around the 2MB mark or anything?  I see a 10MiB limit for JSON API calls but don't think we'd quite be hitting that:

https://cloud.google.com/storage/quotas

I also tried locally and it has the GCP storage I think that corresponds to limits above or uses in memory storage or uses SQL-lite DB (the default when running locally)

https://www.sqlite.org/limits.html

I tried with your example locally with the SQL-Lite DB it doesn't seem to run into any issues with the OpenCircuits API itself here or SQL-Lite, but I'm guessing it is GCP or backend storage account related so maybe not something I'd be able to replicate.

Long story short, if run it all local from v3.0 branch/tag then it seems to work fine so maybe can self host if it's a major annoyance, but isn't just you :)

Also first time posting here but love the project it is a lot of fun.

shusain, 2023-10-08:
Oh also I think this should be labelled backend instead of frontend probably because it gives a 500 error in the network requests from the Frontend to the Backend (internal server error that happened on the server trying to save it rather than FE error I think)

LeonMontealegre, 2023-10-08:
Hey! Thanks for reporting this, we're actually currently (well kinda on hold atm) working on a huge refactor that will dramatically reduce circuit sizes and I'll add this to our backlog to also look at the GCP limits and such. But yeah as @shusain mentioned, you can always locally host things which should work in the meantime.
I appreciate the first-posts, engagement is always nice!
Also shameless plug for our discord (https://discord.gg/bCV2tYFer9) if either of you would like to join and are interested in discussing/contributing to OpenCircuits in general :) !